swift-nav/piksi_tools #19: Move console into its own repo?

mfine, 2015-04-15:
Its in its own directory, but does the console even belong in here? Should we just setup a `piksi_console` repo and dump it in there?

/cc @henryhallam @fnoble @mookerji @cbeighley @denniszollo

cbeighley, 2015-04-23:
I'd vote no - Piksi Console is the quintessential Piksi tool. It makes it more granular but without any obvious benefit IMO.

henryhallam, 2015-04-24:
Yeah there is such a thing as having too many repositories.  I do think it's a good idea to separate the installation and dependencies though, because there are circumstances where you'd want to install the lighter-weight command-line tools on e.g. a headless system without bringing in Chaco, QT4 and all that jazz.

closed 2016-08-02 by denniszollo