swift-nav/piksi_tools #39: Bootload client should repeatedly try to reset the Piksi

henryhallam, 2015-05-05:
Presently the client flow is like this:
1. Send SBP_MSG_RESET
2. Wait indefinitely for SBP_MSG_BOOTLOADER_HANDSHAKE
3. If we receive it, reply with another HANDSHAKE and then begin the bootload process.

We should instead:
1. Have a reasonable timeout rather than waiting forever,
2. Send SBP_MSG_RESET repeatedly during that timeout.

This avoids the situation where the Piksi happens to be still booting up when we first send SBP_MSG_RESET, that message never gets received, and the client waits forever for the bootloader to appear while printing all the usual app SBP_MSG_PRINTs.

/cc @fnoble @denniszollo

henryhallam, 2015-07-02:
FYI @cbeighley

cbeighley, 2015-07-02:
@henryhallam Mark has added this: https://github.com/swift-nav/piksi_tools/blob/master/piksi_tools/bootload.py. Okay to close?

henryhallam, 2015-07-03:
Thanks @mfine !

closed 2015-07-03 by henryhallam