swift-nav/piksi_tools #40: Update wiki to reflect piksi_firmware/scripts move to piksi_tools

mfine, 2015-05-06:
There are a couple of pages on the wiki (and a video) still referencing `piksi_firmware/scripts`:

http://docs.swift-nav.com/wiki/HOW-TO:_Flashing_the_Piksi_firmware_using_the_bootloader
http://docs.swift-nav.com/wiki/HOW-TO:_Running_the_Piksi_Console_from_source

Documentation should also reflect moving the console into its own directory at `piksi_tools/console` and running it from the source.

/cc @denniszollo

denniszollo, 2015-05-06:
I set up these wiki documents to refer to piksi_tools.  I'm testing a bit now to make sure that we aren't missing any requirements.

mfine, 2015-05-07:
Documentation has been updated and there's a new setup.sh script.

closed 2015-05-07 by mfine