swift-nav/piksi_tools #75: Running piksi_tools from source options are confusing

cbeighley, 2015-06-22:
http://docs.swiftnav.com/wiki/HOW-TO:_Running_the_Piksi_Console_from_source

It's unclear if one should install ```piksi_tools``` from ```pip```, or download ```piksi_tools``` repo and run from source. When running from source, should you add ```piksi_tools``` path to ```PYTHONPATH```? Should you do this in your ```~/.bashrc```? We should have clearer documentation around what the recommended way to run from source is.

Also, should we build Ubuntu console binaries and distribute them with releases the same way we do for Windows and OSX?

mfine, 2015-06-22:
This page is pretty explicit about running from the source repo - there's no mention about pip installing piksi_tools. Maybe the documentation here just needs some refresh and updating - the paths to the console are wrong and don't know if out of the box its necessary to set your PYTHONPATH.

mfine, 2015-06-22:
Just ran through these steps in a `14.04` container. Think we should dump the `PYTHONPATH` instructions and instruct people to install the package with `python setup.py install`. That worked well for me. We should consider the PYTHONPATH stuff a hack.

What do you guys think? @cbeighley @denniszollo Thinks all we need here to to remove the optional install bit and update the path and we're all set.

mfine, 2015-06-22:
Commands run in a container that worked:

```bash
$ apt-get install git
$ git clone https://github.com/swift-nav/piksi_tools.git
$ cd piksi_tools/
$ ./setup.sh -x install
$ pip install -r requirements.txt
$ python setup.py install
$ python piksi_tools/console/console.py
```

this also worked, with the PYTHONPATH:

```bash
apt-get install git
git clone https://github.com/swift-nav/piksi_tools.git
cd piksi_tools/
./setup.sh -x install
pip install -r requirements.txt
PYTHONPATH=. python piksi_tools/console/console.py
```

mfine, 2015-06-22:
Updated the wiki with edited instructions. Also setup a Dockerfile for running piksi_tools as a container. Not sure how well that works with gui tools.

closed 2015-06-22 by mfine