swift-nav/piksi_tools #109: Settings: default vals and enum options are duplicated between firmware code and yaml

henryhallam, 2015-07-08:
Default values for settings, and also the string representations of possible enum values, are defined both in piksi_firmware and in settings.yaml.  This presents the possibility for conflicts when someone only remembers to update them in one place.

Not sure what the best approach is here; proposed solutions invited.

@denniszollo @fnoble @mfine @mookerji

fnoble, 2015-07-08:
Could we just remove them? Are they currently just used for the PDF documentation?

henryhallam, 2015-07-08:
Also for the console.  Maybe we could provide a means to read them out of the firmware, but that would make the doc generation more annoying.

denniszollo, 2015-07-08:
I really like seeing the default values in the console.  It's valuable to
know whether you have the default settings or not, especially prior to
hitting the "reset to defaults" button.  If we read them out the firmware,
it would still be hard to make sure that they are correct.

Maybe we should extend the settings read message to send the current and
default value.