swift-nav/piksi_tools #131: Baseline plotting bug

mookerji, 2015-07-22:
Came up while testing v0.19-rc0-52 today.

[image]

mfine, 2015-07-22:
@henryhallam I walked this down to:

https://github.com/swift-nav/piksi_tools/compare/c802825dc8cd73a10914024b7d8da68974f854d2...80e1c32109608e9bf5b58c64d9359164e8a3cef8

Traces in my console:

```
ValueError: All-NaN slice encountered
ERROR:traits:Exception occurred in traits notification handler for object: <chaco.array_plot_data.ArrayPlotData object at 0x11d1794d0>, trait: data_changed, old value: <undefined>, new value: {'changed': ['n_float']}
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/usr/local/lib/python2.7/site-packages/chaco/plot.py", line 1061, in _data_update_handler
    source.set_data(self.data.get_data(name))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 98, in set_data
    self._compute_bounds()
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 253, in _compute_bounds
    self._min_index = bounded_nanargmin(data.view(ndarray))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 14, in bounded_nanargmin
    min = nanargmin(arr)
  File "/usr/local/lib/python2.7/site-packages/numpy/lib/nanfunctions.py", line 375, in nanargmin
    raise ValueError("All-NaN slice encountered")
ValueError: All-NaN slice encountered
ERROR:traits:Exception occurred in traits notification handler for object: <chaco.array_plot_data.ArrayPlotData object at 0x11d1794d0>, trait: data_changed, old value: <undefined>, new value: {'changed': ['e_float']}
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/usr/local/lib/python2.7/site-packages/chaco/plot.py", line 1061, in _data_update_handler
    source.set_data(self.data.get_data(name))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 98, in set_data
    self._compute_bounds()
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 253, in _compute_bounds
    self._min_index = bounded_nanargmin(data.view(ndarray))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 14, in bounded_nanargmin
    min = nanargmin(arr)
  File "/usr/local/lib/python2.7/site-packages/numpy/lib/nanfunctions.py", line 375, in nanargmin
    raise ValueError("All-NaN slice encountered")
ValueError: All-NaN slice encountered
ERROR:traits:Exception occurred in traits notification handler for object: <chaco.array_plot_data.ArrayPlotData object at 0x11d1794d0>, trait: data_changed, old value: <undefined>, new value: {'changed': ['n_float']}
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 520, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/usr/local/lib/python2.7/site-packages/traits/trait_notifiers.py", line 483, in dispatch
    handler( *args )
  File "/usr/local/lib/python2.7/site-packages/chaco/plot.py", line 1061, in _data_update_handler
    source.set_data(self.data.get_data(name))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 98, in set_data
    self._compute_bounds()
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 253, in _compute_bounds
    self._min_index = bounded_nanargmin(data.view(ndarray))
  File "/usr/local/lib/python2.7/site-packages/chaco/array_data_source.py", line 14, in bounded_nanargmin
    min = nanargmin(arr)
  File "/usr/local/lib/python2.7/site-packages/numpy/lib/nanfunctions.py", line 375, in nanargmin
    raise ValueError("All-NaN slice encountered")
ValueError: All-NaN slice encountered
```

henryhallam, 2015-07-22:
> c802825...80e1c32

Thanks! I'll look into this while Colin drives in circles for the next 30
mins :)

mfine, 2015-07-22:
Verified that those arrays are going in with all nans for floats and are not being updated without nans. Not sure the intention here, nor can I really follow what's going on with `self.neds` in `BaselineView` - but an array of all nans into `nanargmin` is going to eek per the [doc](http://docs.scipy.org/doc/numpy/reference/generated/numpy.nanargmin.html). Not sure what the correction here should be.

closed 2015-07-24 by denniszollo

henryhallam, 2015-07-24:
> Closed #131 <https://github.com/swift-nav/piksi_tools/issues/131> via #138

Thanks for cleaning up my mess!

PR #138 (2015-07-24): baseline: Fix barfing on NaNs when there's no data.
Fixes #131