swift-nav/piksi_tools #144: Set position from mean

Matioupi, 2015-07-27:
It would be nice to have a button/funcion in the console to set (fix) the base station position from the mean single point solution of the last N seconds

jkretzmer, 2016-06-20:
A quick note here - when implementing this feature, it may be wise to indicate that Piksi was operating in self-survey mode to obtain this base position. That way, any downstream consumers of this data are aware that the survey position is probably not of particularly high quality / fidelity.

I'm guessing the relevant SBP messages are MSG_BASE_POS_ECEF and MSG_BASE_POS_LLH. We may want to consider adding a status flags field similar to the one found in MSG_POS_ECEF and MSG_POS_LLH. In those messages, flags are used to indicate what the type of solution is (RTK Fix, RTK Float, SPP). In the BASE messages, flags could be used to distinguish how the base station position was acquired (some ideas for these types: Self Survey, Survey Marker, Manual Input from External GPS).

denniszollo, 2016-08-02:
@Matioupi.  We've implemented this feature with PR #303.   We hope to release with a v0.28 of the console.  Thank you so much for the constructive feedback.

Matioupi, 2016-08-03:
Hello,

thanks for the info.

do not forget to provide an easy (automatic) way to output / save the set position for future reference, eg translating a set of surveyed point at mobile established from "mean fake survey position at base" when the true base survey position may be known accurately later on (post-processing or other means)

regards,

Mathieu