swift-nav/piksi_tools #150: Add button to enable/disable CSV logging

cbeighley, 2015-07-30:


switanis, 2016-05-26:
All logging should be off by default when Console is started. Might be a command line option to enable logging immediately though.

Allow to start and stop logging when console is opened. Allow to select folder for logging. Do not log anything unless commanded to. Indicate logging status on GUI (eg. star in a title bar, folder/file name in status bar). Use GUI buttons to start and stop logging. 
Windows default folder: c:\Users\<username>\Documents\Piksi-Logs\
Memorize last used folder and use as default for the next session.

jacobmcnamee, 2016-05-26:
I feel like every time I've wanted a log was after the fact. What is the issue with logging by default? Though I agree the folder should change.

switanis, 2016-05-26:
Can be a program option ("Enable logging automatically after start") and also as a command line option.
I don't like unwanted stuff on my PC. And there is no logging place selected after start yet. 

Such option (Enable Logging Now) can also be a part of a startup popup window where a COM port is selected. With default setting memorized from previous choice.

closed 2016-08-02 by denniszollo