swift-nav/piksi_tools #167: Update documentation for console changes

denniszollo, 2015-08-21:
We will need to replace or crop screenshots that show the message window or the update screen before we release a new version of the console.

* Clever Idea is to so some image manipulation and just replace the bottom half of the screen on all the screenshots with the new looking msg table.  
* Firmware update instructions will need some love as well as this screen was reshuffled.

@robhranac, @drewshannon

denniszollo, 2015-08-21:
New console looks like this:

[image]

denniszollo, 2015-08-21:
[image]

sashamitelman, 2015-10-16:
I added my own suggestions in #182.

sashamitelman, 2015-10-16:
I didn't see this initially, so I added my own suggestions in #182.

denniszollo, 2015-11-30:
@drewshannon , @rai-swiftnav , @robhranac 
As the documentation team, we need to audit documentation for compatibility with the new look of the firmware update team.

closed 2021-10-13 by silverjam