swift-nav/piksi_tools #182: Piksi User Getting Started Guide proposed edits

sashamitelman, 2015-10-16:
* Unclear whether or not it is safe to hot-plug the Piksi to the host PC (USB) and/or the 3DR 915 MHz radios. Would be good to state this in the instructions, especially if there's any risk of damaging the electronics by doing so.

* Under the "Viewing position solutions" section, the text describing the tracking tab states "This simulation shows that you are tracking 8 satellites; you know this because you will see consistent C/No of over 33 dB-Hz." The corresponding screenshot shows 9SVs. Also, the statement "consistent C/N0 of over 33 dBHz" does not necessarily follow from the number of SVs tracked; is this just stating our acquisition threshold?

* The screenshot for "Baseline tab" and the accompanying text description differ from the output of the current Console build. ("Rover path" and "Rover Relative Position" are replaced with "RTK Fixed" and "RTK Float," respectively, and the blue trace is not shown in the current build's demo run.)

denniszollo, 2015-11-30:
@drewshannon for the next video rev

robhranac, 2015-12-08:
We're going to no op para 1.  Paragraph 2 is now updated.  Paragraph 3 is problematic, so I am keeping this open.  All others are video revs that are no ops for now.

sashamitelman, 2015-12-09:
May I suggest adding a few Amadana-style illustrations to the antenna
placement guidelines?

http://gizmodo.com/339479/amadanas-warning-illustrations-are-something-special

closed 2021-10-13 by silverjam