swift-nav/piksi_tools #209: A Few Missing Dependencies.

akleeman, 2015-11-19:
Had some issues installing piksi_tools.  I started by following the [piksi docs](http://docs.swiftnav.com/wiki/HOW-TO:_Running_the_Piksi_Console_from_source), then switched to the readme in piksi_tools (which both have slightly different suggestions for install).  It seems that both are missing a few required packages.  Namely:  `pyqt`, `chaco`, `traits` and `traitsui`.

(Note: installed it using a conda environment on ubuntu)

mookerji, 2015-11-19:
@akleeman Did you use the Makefile in the repo?

akleeman, 2015-11-19:
Yeah, I checked out the rep, ran:
```
make deps
make serial_deps
python setup.py install
```

akleeman, 2015-11-19:
The problem was that I was attempting to install this in a virtualenv, not using the system python, so anything installed via apt-get wasn't being linked to my virtual python.

Could be worth moving dependencies over to setup.py instead of the shell script (which would make it work on both system and virtual python environments), but that's likely not high priority!

closed 2015-11-19 by akleeman