swift-nav/piksi_tools #269: Console: Firmware update always require to always select a file before updating STM

switanis, 2016-05-11:
Console v0.26. 
Before pressing on Update STM button it is always required to browse and select a file. If the file is selected once and it has changed in the background (eg. after new compilation) Console will use the previously selected file and not the new one as expected.

ljbade, 2016-05-17:
Yeah I have noticed this before. I always reselect the file each time to make sure it gets the new version.

ljbade, 2016-05-17:
@denniszollo Would this be an easy fix? Always read the file?

denniszollo, 2016-05-19:
Not an easy fix, but we can consider.  IMO the fix is to clear the file from memory in the console as soon as you bootload so at least it's not confusing.

ljbade, 2016-05-23:
Yes that would also work