swift-nav/piksi_tools #278: Add status bar

switanis, 2016-05-26:
It will be very helpful to display a status bar at the bottom of the Console window. It should contain:

COM port closed/ COMx port opened status. Current data rate Rx and Tx in KB/s. May show rotating bar - \ | / - ... when messages are arriving as well.

Piksi serial number.

Logging status (on/off), logging folder, current file size (xx B, xx.x KB, xx.xx MB, xx.xxx GB).

Solution status: Receiver off, Antenna open/short, No Signal, Tracking, SPS, Float RTK, Fixed RTK. Use field background color for clear identification: red (error, no data, no RF), cyan or magenta (tracking), blue (SPS), yellow or orange (float), green (fixed).

Number of used and tracked SVs.

Use font large enough for easy reading in a moving vehicle.

denniszollo, 2016-06-16:
We should draw a little mock-up as a starting point.  I think this would add a tone of value in the console.

denniszollo, 2016-06-17:
[image]

denniszollo, 2016-08-02:
done with #301

closed 2016-08-03 by asthakeshan