swift-nav/piksi_tools #320: Some anti-virus softwares do not like the console.exe

denniszollo, 2016-08-08:
AV complains about .exe (same for 0.27)
[image]

Investigate methods to fix.  Do we need to publish the software with microsoft or some kind of authority?

ljbade, 2016-08-09:
You need to sign the exe

ljbade, 2016-08-09:
https://msdn.microsoft.com/en-us/library/ms537361(v=vs.85).aspx
https://msdn.microsoft.com/en-us/library/windows/desktop/aa388170(v=vs.85).aspx

The first step is to get a code signing certificate.

We could also do Mac OSX code signing: https://developer.apple.com/library/mac/documentation/Security/Conceptual/CodeSigningGuide/Introduction/Introduction.html

cbeighley, 2016-12-06:
Create with later version of InstallShield.

ljbade, 2016-12-06:
@cbeighley Does later version create a signed exe?