swift-nav/piksi_tools #348: Ubuntu 16.04 needs python-wxgtk30

ljbade, 2016-10-18:
Latest Ubuntu I have (16.04) needs python-wxgtk30 in place of python-wxgtk28 as the older package has been removed. This means the setup script fails on 16.04.

I have not tested 16.10.

/cc @denniszollo

ljbade, 2016-10-18:
Another problem with 16.04, the latest pip package is broken, and has a bug that results in `AttributeError: 'Requirement' object has no attribute 'project_name'` error when running the requirements.txt

There are several ways around it:
- upgrade pip using `pip install --upgrade pip`
- downgrade the apt package to `python-pip=8.1.1-2`

ljbade, 2016-10-18:
we might want to add to the install script a check for the buggy pip package verson `8.1.1-2ubuntu0.1` and tell user to upgrade pip, or do it for them

ljbade, 2016-10-18:
Looks like Ubuntu is going to release https://launchpad.net/ubuntu/+source/python-pip/8.1.1-2ubuntu0.3 to fix this bug

ljbade, 2016-10-18:
Hmm now I can't get enable to install from requirements_gui.txt:

```
Collecting enable==4.4.1 (from -r requirements_gui.txt (line 3))
  Downloading enable-4.4.1.tar.gz (2.2MB)
    100% |████████████████████████████████| 2.3MB 562kB/s 
    Complete output from command python setup.py egg_info:
    /usr/local/lib/python2.7/dist-packages/numpy/distutils/system_info.py:635: UserWarning: Specified path /usr/local/include/python2.7 is invalid.
      warnings.warn('Specified path %s is invalid.' % d)
    /usr/local/lib/python2.7/dist-packages/numpy/distutils/system_info.py:635: UserWarning: Specified path  is invalid.
      warnings.warn('Specified path %s is invalid.' % d)
    non-existing path in 'kiva': 'tests'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-55puGj/enable/setup.py", line 57, in <module>
        config = configuration().todict()
      File "/tmp/pip-build-55puGj/enable/setup.py", line 49, in configuration
        config.add_subpackage('kiva')
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 1002, in add_subpackage
        caller_level = 2)
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 971, in get_subpackage
        caller_level = caller_level + 1)
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 908, in _get_configuration_from_setup_py
        config = setup_module.configuration(*args)
      File "kiva/setup.py", line 27, in configuration
        config.add_subpackage('agg')
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 1002, in add_subpackage
        caller_level = 2)
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 971, in get_subpackage
        caller_level = caller_level + 1)
      File "/usr/local/lib/python2.7/dist-packages/numpy/distutils/misc_util.py", line 908, in _get_configuration_from_setup_py
        config = setup_module.configuration(*args)
      File "kiva/agg/setup.py", line 174, in configuration
        if int(m.group(1)) < 4:
    AttributeError: 'NoneType' object has no attribute 'group'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-55puGj/enable/
```

ljbade, 2016-10-18:
using `enable==4.5.1` seems to fix the issue

ljbade, 2016-10-18:
Now I get another error when I run the console:
```
eith@leith-xps:~/swift/piksi_tools$ python piksi_tools/console/console.py 
Traceback (most recent call last):
  File "piksi_tools/console/console.py", line 13, in <module>
    import piksi_tools.serial_link as s
  File "/usr/local/lib/python2.7/dist-packages/piksi_tools-0.29.post39_gd332784_dirty-py2.7.egg/piksi_tools/serial_link.py", line 23, in <module>
    from sbp.bootload                       import *
  File "/usr/local/lib/python2.7/dist-packages/sbp/bootload.py", line 23, in <module>
    from sbp.msg import SBP, SENDER_ID
  File "/usr/local/lib/python2.7/dist-packages/sbp/msg.py", line 66, in <module>
    class SBP(object):
  File "/usr/local/lib/python2.7/dist-packages/sbp/msg.py", line 72, in SBP
    ULInt8('preamble'),
NameError: name 'ULInt8' is not defined
```

ljbade, 2016-10-18:
It seems Ubuntu 16.04 ships with Construct 2.8 which has major changes from 2.5 (see https://construct.readthedocs.io/en/latest/).

Unfortunately libsbp uses the older 2.5 API. We will need to upgrade libsbp to 2.8

ljbade, 2016-10-18:
We also need to remove distribute, and use setuptools instead, distribute is not working for me in Anaconda, their page https://pythonhosted.org/distribute/ says its no longer maintained.

ljbade, 2016-10-18:
It seems Ubuntu 16.04 does come with construct 2.5, but no idea why it isn't using it (http://packages.ubuntu.com/xenial/python-construct)

ljbade, 2016-10-18:
Thanks to @jkretzmer there is instructions on the forum about these issues: https://groups.google.com/forum/#!searchin/swiftnav-discuss/console$20linux%7Csort:date/swiftnav-discuss/aYEYZ70N_Yg/VodGK7EVBAAJ

We should try to get these in.

ljbade, 2016-10-18:
Going to start making some of these improvements in https://github.com/swift-nav/piksi_tools/pull/349

PaulBouchier, 2017-02-05:
Thanks for the link to jkretzmer instructions. They worked for me on 16.04.1