swift-nav/piksi_tools #393: Final version of Piksi v2 console and warning pop-ups

cbeighley, 2016-12-13:
We're going to release a final version of the old console (for Piksi v2), that will pop up a warning message if it sees that you are using a Piksi Multi with it, telling you to upgrade your console to the latest version for Piksi Multi. Dennis to decide where this gets forked off (pre Piksi Multi upgrade screen?).

The new console will work for v2 (although we are specifically _NOT_ going to test it), but if used with a Piksi v2, there will be a pop-up telling you to use the last Piksi v2 console release. The new console will have a different place in the `index.json` on downloads.swiftnav.com to go look for new consoles.

denniszollo, 2016-12-14:
I'm actually thinking now that we will rename the NEW console to "Swift Console", so it is forward compatible with any products we have that are not named Piksi.

denniszollo, 2016-12-15:
Implemented here: https://github.com/swift-nav/piksi_tools/tree/last_official_v2_console

TODO: make release binaries; send message to forum explaining what we did

denniszollo, 2017-01-16:
this is finally released and sent to forum

closed 2017-01-16 by denniszollo