swift-nav/piksi_tools #403: Unexpected tracking channel legend behavior

wltr, 2016-12-17:
The tracking legend seems to have duplicates and not display tracking channel use correctly.

[image]
[image]

denniszollo, 2016-12-22:
there is a new legend entry for items that are dropped on one channel and come back on a different channel.  The legend entry doesn't go away until that item scrolls out of the history (200 tracking interations later).  

How do you think it should work?

denniszollo, 2016-12-22:
Ok, I agree that it should just have the most recent channel in the legend.  But I do think your tracking performance must have been really bad in order to see a legend that long.

PR #412 (2016-12-23): only put legend entries for currently tracked items
should fix #403
closed 2016-12-23 by denniszollo