swift-nav/piksi_tools #525: ImportError: No module named gui_utils

rubenmp10, 2017-03-21:
Hi, 

I have followed the steps shown in http://docs.swiftnav.com/wiki/HOW-TO:_Running_the_Piksi_Console_from_source to install the console on my Ubuntu 14.04.

When I try to run it using 
python piksi_tools/console/console.py
I get the following error:

git version 1.9.1
WARNING:traits.has_traits:DEPRECATED: traits.has_traits.wrapped_class, 'the 'implements' class advisor has been deprecated. Use the 'provides' class decorator.
Traceback (most recent call last):
  File "piksi_tools/console/console.py", line 138, in <module>
    from piksi_tools.console.solution_view import SolutionView
  File "/usr/local/lib/python2.7/dist-packages/piksi_tools-1.0.0.post113_gad2320f_dirty-py2.7.egg/piksi_tools/console/solution_view.py", line 20, in <module>
    from piksi_tools.console.gui_utils import plot_square_axes, MultilineTextEditor
ImportError: No module named gui_utils

Does anybody know what I should do to fix the error?

Thanks, 
Ruben

denniszollo, 2017-03-31:
Ruben,

I believe if you checkout latest master this is resolved.

Also, we provide packaged versions for Ubuntu because running from source is not very user friendly:

You should be able to untar this file linked to on the page below and run the console executable.  
http://support.swiftnav.com/customer/en/portal/articles/2492795-swift-console-piksi-console-

rubenmp10, 2017-04-06:
Thank you Dennis, I downloaded the file you mentioned and now I can run the console executable without any problems.

closed 2017-04-20 by denniszollo