swift-nav/piksi_tools #615: console shows PRN+1 instead of PRN

mbavaro, 2017-07-11:
The new tracking state message uses `gnss_signal16_t` where the PRN is already 1-based. I think the console is adding 1 to that (conforming with the previous `MSG_TRACKING_STATE`)

mbavaro, 2017-07-11:
in case this is the right thing to do (I am fairly convinced it is), here is the PR:
https://github.com/swift-nav/piksi_tools/pull/616

Paakkit, 2017-07-12:
Indeed we have thing like PRN33 appearing in console.

[image]

ljbade, 2017-07-12:
Wow this has been an issue for a while I believe, nice work!

mbavaro, 2017-07-12:
Keerthan will embrace this change in the next PR, closing as we verify all is gone well!

commit 1a87c815 (2017-07-27): tracking_view: don't increment gps prn

fixes #615
closed 2017-07-27 by jck