swift-nav/piksi_tools #691: Console crashes with MemoryError after recording JSON log for 5-8 hours (Windows 10)

nsirola, 2017-12-18:
I used to be able to log data over the weekend with the console, but the last couple of versions crash during the first hours with:

Exception occurred in traits notification handler for object: <chaco.array_data_source.ArrayDataSource object at 0x092811E0>, trait: data_changed, old value: <undefined>, new value: True
Traceback (most recent call last):
  File "site-packages\traits\trait_notifiers.py", line 519, in _dispatch_change_event
  File "site-packages\traits\trait_notifiers.py", line 482, in dispatch
  File "site-packages\chaco\base_xy_plot.py", line 636, in _either_data_changed
  File "site-packages\enable\component.py", line 562, in invalidate_draw
  File "site-packages\enable\component.py", line 565, in invalidate_draw
  File "site-packages\enable\abstract_window.py", line 235, in invalidate_draw
MemoryError

[image]

The console is responsive and lets me switch tabs, but data does not update although the Piksi is running. 
Console takes 2590 MB memory at this point. Since it starts out using around 85MB, there seems to be a memory leak somewhere.

nsirola, 2017-12-18:
Actually the last successful over-the-weekend log was in February, with the console version that saved files named `serial-link-*` instead of `swift-gnss-*`

The longest I've been able to log since fall seems to be 34 hours (3.1GB log) on Nov 10th,  otherwise the over-the-weekend logs seem to have collected only the first 5 to 15 hours of data.

nsirola, 2017-12-19:
This only seems to happen when on the Solution or Observartions tab. After a night logging JSON with the tracking tab open, the console is still running and taking less than 100MB memory.

mbavaro, 2017-12-21:
I am using 1.3.7, and also see slowly increasing memory usage
[image]
I don't think which tab is ON has anything to do with the leak.

nsirola, 2017-12-22:
I opened two instances of 1.3.7 console in Windows 10. I left one (COM4) on the Tracking tab, and switched the other (COM8) to the Solution tab and then did not touch them for an hour or so. The memory use of the Tracking one fluctuates between 40-80MB, while the Solution one keeps steadily growing and is at 366MB after one hour:

[image]

pmiettinen, 2017-12-22:
Does the memory consumption drop if you press clear history button?

nsirola, 2017-12-22:
No

pmiettinen, 2017-12-22:
Nothing really obvious pops up in the `solution_view.py`. Probably will need somekind of memory profiling tool. One note is that appending tables and calling `update_table` should be protected with thread-safety features.

jaredw42, 2019-06-22:
This should be fixed with @tn-exafore's refactor of the console.  Closing

closed 2019-06-22 by jaredw42