swift-nav/piksi_tools #1078: clang: error: unknown argument: '-framework ApplicationServices'

makslevental, 2019-06-19:
also

`clang: error: unknown argument: '-framework OpenGL'`

somehow related to this?

https://github.com/enthought/enable/pull/335

please advise @silverjam

silverjam, 2019-06-20:
Hello @makslevental, it could be related, can you describe where you're seeing the error?  Also, if possible I would use a released version of the console from https://support.swiftnav.com/customer/en/portal/articles/2492795-swift-console -- however, the version of the `enable` library that the console builds against *should* have this issue fixed.

silverjam, 2019-06-21:
Hi @makslevental -- as I mentioned on another ticket, please open a support request through support@swiftnav.com if the released version of the console doesn't work out (it definitely should though!).

closed 2019-06-21 by silverjam