swift-nav/piksi_tools #1122: Linux GUI installation process missing "enable" on Ubuntu 18.04

stoooops, 2020-01-20:
Hello, I am having trouble installing and running the console GUI on Ubuntu 18.04

Here are the commands I am using to install (assuming pyenv is installed for virtualenv but equivalent if you use other tooling)
```
REPO_NAME=piksi_tools

# clone repo
git clone git@github.com:swift-nav/${REPO_NAME}.git
cd ${REPO_NAME}/

# setup virtualenv your favorite way
pyenv virtualenv 3.5.8 ${REPO_NAME}
echo "${REPO_NAME}" > .python-version

# install
make deps
pip install -e .

# run
python -m piksi_tools.console.console
```
I get:
```
Traceback (most recent call last):
  File "/home/coryg/.pyenv/versions/3.5.8/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/coryg/.pyenv/versions/3.5.8/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/coryg/code/piksi_tools/piksi_tools/console/console.py", line 25, in <module>
    from enable.savage.trait_defs.ui.svg_button import SVGButton
ImportError: No module named 'enable'
```

I actually remember encountering this a few months ago using your software, and then I fixed it. But I can't remember how I did it. I think there is a dependency issue.

stoooops, 2020-01-20:
If I try to fix it by running
```
> pip install enable
...
Successfully installed enable-4.8.1
```

I get:
```
(piksi_tools) coryg@seedy:~/code/piksi_tools$ python -m piksi_tools.console.console
Traceback (most recent call last):
  File "/home/coryg/.pyenv/versions/piksi_tools/lib/python3.5/site-packages/enable/savage/trait_defs/ui/toolkit.py", line 33, in _init_toolkit
    __import__(backend)
ImportError: No module named 'enable.savage.trait_defs.ui.null'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/coryg/.pyenv/versions/3.5.8/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/coryg/.pyenv/versions/3.5.8/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/coryg/code/piksi_tools/piksi_tools/console/console.py", line 25, in <module>
    from enable.savage.trait_defs.ui.svg_button import SVGButton
  File "/home/coryg/.pyenv/versions/piksi_tools/lib/python3.5/site-packages/enable/savage/trait_defs/ui/svg_button.py", line 19, in <module>
    from .svg_button_editor import SVGButtonEditor
  File "/home/coryg/.pyenv/versions/piksi_tools/lib/python3.5/site-packages/enable/savage/trait_defs/ui/svg_button_editor.py", line 24, in <module>
    from enable.savage.trait_defs.ui.toolkit import toolkit_object
  File "/home/coryg/.pyenv/versions/piksi_tools/lib/python3.5/site-packages/enable/savage/trait_defs/ui/toolkit.py", line 43, in <module>
    _init_toolkit()
  File "/home/coryg/.pyenv/versions/piksi_tools/lib/python3.5/site-packages/enable/savage/trait_defs/ui/toolkit.py", line 36, in _init_toolkit
    "toolkit." % ETSConfig.toolkit)
ImportError: Unable to import a Savage backend for the null toolkit.
```

stoooops, 2020-01-20:
These issues appear related, but were closed:
https://github.com/swift-nav/piksi_tools/issues/1071
https://github.com/swift-nav/piksi_tools/issues/1080

stoooops, 2020-01-20:
This related issue from another repo as well: https://github.com/enthought/enable/issues/162

stoooops, 2020-01-20:
It appears the `make deps` does not actually install the GUI requirements, which the README says it does.

If I additionally run
```
pip install -r requirements_gui.txt
```
then it will install `enable-4.7.1.dev68` but I still get the same error: `ImportError: Unable to import a Savage backend for the null toolkit.` (with same stack trace pasted above)

silverjam, 2020-02-06:
Hi @stoooops -- sorry the docs are out of date in current areas at the moment.  If you don't need to modify the console, I'd recommend running a pre-built version of it for Ubuntu 18.04 here:

https://console-ci-builds.s3-us-west-2.amazonaws.com/v2.3.21/linux_ubuntu_1804/swift_console_v2.3.21_ubuntu1804.tar.gz

To run this simple cd into the extract directory and run `./console`.

If you need to run from source I recommend using Miniconda3 to generate a Python environment:
```
conda create -n piksi_tools py=3.7
conda activate piksi_tools
```

The current best "source of truth" for Python dependencies is the [tox.ini][tox_ini] file that builds the binary distribution.

[tox_ini]: https://github.com/swift-nav/piksi_tools/blob/v2.3.0-release/tox.ini#L48

Which looks something like this when you run it:
```
pip install -r requirements.txt
pip install pyqt5==5.10.0
pip install -r requirements_gui.txt
pip install -e .
```

Then you can run:
```
python -m piksi_tools.console <...args...>
```

...to launch the console.  Also please note you can contact our support email (support AT swiftnav.com) if you need a quicker turnaround on issues like this.

PR #1128 (2020-02-06): update docs on running the console from source
Fixes #1122
closed 2020-02-06 by silverjam