swift-nav/piksi_tools #1223: Big Sur breaks pyserial 3.4...

denisng, 2021-02-18:
This happens on macOS Big Sur v11.2 on branch `v3.0.0-relesae` (as of commit 790e77769669ed94de5069d9e7c2e7299585d310):

```
$> python serial_link.py -h
                                                                                                                                                                                                                                                      
Traceback (most recent call last):
  File "serial_link.py", line 22, in <module>
    import serial.tools.list_ports
  File "/Users/denisng/miniconda3/envs/piksi_tools/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/denisng/miniconda3/envs/piksi_tools/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/denisng/miniconda3/envs/piksi_tools/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
```

Issue appears to be fixed in pyserial-3.5

silverjam, 2021-10-13:
@rai-swiftnav Can you check and see if you see the same thing on Big Sur, does a serial connection to a Piksi work with console 3.1.4?

switanis, 2021-10-13:
I see no console issues on Big Sur mac v11.6 over UART with two different USB-RS232 adapters)

silverjam, 2021-10-14:
Going to close this for now, @denisng please re-open (or open an internal issue) if the problem reoccurs

closed 2021-10-14 by silverjam