swift-nav/piksi_tools #1228: mac gui dep enthought/enable failed to build on py35

jayvdb, 2021-03-11:
While running `make deps` on 11.2.3 (20D91)
```
build/src.macosx-10.6-x86_64-3.5/kiva/quartz/ABCGI.c:19545:49: error: implicit declaration of function 'CGGLContextCreate' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
      __pyx_v_self->__pyx_base.__pyx_base.context = CGGLContextCreate(((void *)__pyx_v_glcontext), CGSizeMake(__pyx_v_width, __pyx_v_height), NULL);
                                                    ^
    build/src.macosx-10.6-x86_64-3.5/kiva/quartz/ABCGI.c:19545:49: note: did you mean 'CGPDFContextCreate'?
```
I am still fumbling around a bit on Mac.  I tried to find that in the SDK headers of Xcode (and Xcode command line tools), to no avail.

I was able to get it building by using Python 3.8 and latest deps, with

```diff
diff --git a/requirements.txt b/requirements.txt
index 8c85d8c..ba37543 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -6,7 +6,7 @@ configparser
 future==0.17.1
 futures==3.1.1
 pep8==1.7.0
-pyserial==3.4
+pyserial==3.5
 pylibftdi==0.16.1.2
 pyparsing==2.2.0
 pygments==2.2.0
diff --git a/requirements_gui.txt b/requirements_gui.txt
index 1101caa..8cea30b 100644
--- a/requirements_gui.txt
+++ b/requirements_gui.txt
@@ -1,5 +1,5 @@
-enable==4.8.1
-chaco==4.8.0
-traits==4.6.0 
-traitsui==6.0.0
-pyface==6.0.0
+git+https://github.com/enthought/enable
+chaco
+traits
+traitsui
+pyface
diff --git a/tasks/setup.sh b/tasks/setup.sh
index 7a76437..283e0b9 100755
--- a/tasks/setup.sh
+++ b/tasks/setup.sh
@@ -211,7 +211,7 @@ function install_python_deps_osx () {
     local conda_env_name
     conda_env_name=$(echo "$ROOT" | sed -e "s@${HOME}/@@" -e 's@/@_@g')
 
-    conda create -n "$conda_env_name" python=3.5 --yes
+    conda create -n "$conda_env_name" python=3.8 --yes
     {
       export PS1=''
 
@@ -224,7 +224,7 @@ function install_python_deps_osx () {
     pip install -r "$ROOT/requirements_gui.txt"
     pip install -e "$ROOT"
 
-    pip install PyQt5==5.10.0
+    pip install PyQt5 # ==5.10.0
 
     log_info ""
     log_info "To run piksi_tools from source, do the following:"
```

jayvdb, 2021-03-15:
fwiw, the gui with those modifications core dumped in `sip` when I connected to a relay.

jayvdb, 2021-03-15:
I cherry-picked https://github.com/enthought/enable/commit/f8783c77 onto v4.8.1, and that also fixed the build, but then I had another seg fault in what appears to be a similar spot.

jayvdb, 2021-03-15:
Ran it under PyCharm/pydev, and have a backtrace after injecting `import faulthandler; faulthandler.enable()`

```py
Fatal Python error: Segmentation fault

Thread 0x00007000104fc000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 463 in _handle_results
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000fff9000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 415 in _handle_tasks
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000faf6000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 406 in _handle_workers
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000f5f3000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 108 in worker
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000f0f0000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 108 in worker
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000ebed000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 108 in worker
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 862 in run
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 882 in _bootstrap
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/_pydev_bundle/pydev_monkey.py", line 795 in __call__

Thread 0x000070000e6ea000 (most recent call first):
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/threading.py", line 293 in wait
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/queue.py", line 164 in get
  File "/usr/local/Caskroom/miniconda/base/envs/piksi_tools/lib/python3.5/multiprocessing/pool.py", line 108 in worker
[... cut ...]
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/pydevd.py", line 1494 in _exec
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/pydevd.py", line 1470 in run
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/pydevd.py", line 2158 in main
  File "/Applications/PyCharm CE.app/Contents/plugins/python-ce/helpers/pydev/pydevd.py", line 2167 in <module>

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```

silverjam, 2021-10-13:
I've tried various combinations of PySide/PyQt5 and newer enthought libs, some combinations work well as long as PyInstaller is not involved, at that point PyInstaller does something that breaks PyQt5/PySide2 -- crash seen is the same as https://github.com/marcelotduarte/cx_Freeze/issues/1261.  So, future progress on this issue will probably mean moving away from PyInstaller to a different packaging/distribution technology such as https://github.com/indygreg/python-build-standalone which does require so much mangling/workarounds for large and complicated packages like PySide2/PyQt5.